loicland/superpoint_graph #217: Is the implementation of SSP-Cluster provided?

jibrilmuhammadadam, 2020-07-21:
Hi, 
Thank you very much for providing your work. In Section 3.3 of the paper "Point Cloud Oversegmentation with Graph-Structured Deep Metric Learning" you said you implemented a 3D  generalization method for "Superpixel Sampling Networks". Is the implementation provided here?
Thank you.

loicland, 2020-08-03:
Hi, sorry for delay.

It is not provided in this repo because it is quite different from SPG.

I have my own implementation in a private repo, which is nolonger maintained, and indlcudes a (now obsolete) reimplementation of sparse matrix multiplication.

I can invite you to the private repo if you want to, but  I haven't touched it in 2 years so I don't guarantee it will work immediately.

loic

aabbcco, 2020-08-04:
Hi,
surprised to see a SSN 3d implementation here,is the implementation implemented using pytorch or tensorflow? I have tried o get the vanilla implementation(https://github.com/NVlabs/ssn_superpixels) to run in Ubuntu20 but failed.
Could I have the access to the repo? Just to have some reference and test.
Thanks
aabbcco

jibrilmuhammadadam, 2020-08-04:
> Hi, sorry for delay.
>
> It is not provided in this repo because it is quite different from SPG.
>
> I have my own implementation in a private repo, which is nolonger maintained, and indlcudes a (now [...]

Please invite me to the repo I will really appreciate it because it will help me in my work. Thanks in advance.

jibrilmuhammadadam, 2020-08-04:
> Hi,
> surprised to see a SSN 3d implementation here,is the implementation implemented using pytorch or tensorflow? I have tried o get the vanilla [...]

This link https://github.com/perrying/ssn-pytorch is the pytorch implementation of SSN. Hope it helps.

loicland, 2020-08-04:
I invited both of you to the private repo.

The relevant code is in:
`embedding/cluster_data_loader.py` and `embedding/cluster_embed.py`.

If I remember correctly, the first function when called will create an "efficient" data structure, the second one will train the network.

Note that:
- the network used to embed the points are quite specific to my application, in the sense that they have very small receptive fields
- the results were not very satisfying (the borders between superpoints were very fractal-y)
- a lot of the code deals with efficient manipulation of sparse soft clustering matrices, which may or may not be made completely irrelevant by the latest sparse matrix libraries such as https://github.com/rusty1s/pytorch_sparse
- I haven't touched this code in over 2 years, so no guarantee that it will work easily. I am willing to help though.

jibrilmuhammadadam, 2020-08-04:
Thanks. I will look at it and if I encounter any problem I'll let you know.
Thanks for the help once again.

aabbcco, 2020-08-04:
> I invited both of you to the private repo.
>
> The relevant code is in:
> `embedding/cluster_data_loader.py` and `embedding/cluster_embed.py`.
>
> If I remember correctly, the first function when called [...]

Thanks for your generous help.

sandeepnmenon, 2021-06-09:
Hi @loicland 
I am also looking to see if the cluster based oversegmentation is more useful for me. I do not care about the "**regularity**" metric. Want to see if I get better results for object purity and border recall using the cluster based oversegmentation.

Can you please invite me as well to your 3d ssn implementation.
Thank you

loicland, 2023-06-16:
Hi!

We are releasing a new version of SuperPoint Graph called SuperPoint Transformer (SPT).

https://github.com/drprojects/superpoint_transformer

It is better in any way:

| ✨ SPT in numbers ✨ |
| :---: |
| 📊 **SOTA results:** 76.0 mIoU S3DIS 6-Fold, 63.5 mIoU on KITTI-360 Val, 79.6 mIoU on DALES  |
| 🦋 **212k parameters only!** | 
| ⚡ Trains on S3DIS in **3h on 1 GPU**| 
| ⚡ **Preprocessing is x7 faster than SPG!** |
| 🚀  **Easy install** (no more boost!)|

If you are interested in lightweight, high-performance 3D deep learning, you should check it out. In the meantime, we will finally retire SPG and stop maintaining this repo.

closed 2023-06-16 by loicland

jibrilmuhammadadam, 2023-06-17:
Thank you for the update and for making the new version publicly accessible.