loicland/superpoint_graph #218: problem with partition

medif, 2020-08-17:
Hi Loic
I had issues with partionning dataset with rgb values, Do have any ideas why the pation is made on segments in vertical way
thanks
[image]

loicland, 2020-08-19:
It is a bit weird yes.
- are you using vanilla spg with handcrafted features? If so can you represent the features visually (option 'f' with visualize.py)
- can you represent the cloud with colors (option 'i')
-

medif, 2020-08-19:
Hi thanks for your response I a using Toronto dataset https://github.com/WeikaiTan/Toronto-3D
-yes
[image]
[image]

or this is the original file:
[image]
-yes they are

loicland, 2020-08-20:
I am unfamiliar with this datasets, and unable to manipulate it myself for now.

A few things seems weird to me on the second image you are showing, provided I can parse the image correctly:
- some ground points are pure red (ie purely linear)
- some power lines are pure blue (ie volumetric and vertical)
- wall segments have different geometric features according to their orientation...
- some spurious periodic patterns

This indicates that they is a problem with how these features are computed.

Some leads:
- check that your reader function does output xyz **in that order**, with z the third dimension
- make sure than another field of the cloud isn't accidentally fed to compute_geof

Could you post here your reader function for this datasets? I will try it myself in the next few days.

medif, 2020-08-26:
Hi thanks for your reply, i  fact i didn't change anything in the compute_geo and  i kept  these lines:
xyz,labels= read_ply(data_file)
 geof = libply_c.compute_geof(xyz, target_fea, args.k_nn_geof).astype('float32')
 features = geof
                 geof[:,3] = 2. * geof[:, 3]
this is the prune functin : xyz, rgb, labels, dump = libply_c.prune(xyz.astype('f4'), args.voxel_width, np.zeros(xyz.shape,dtype='u1'),labels.astype('uint8'),np.zeros(1,dtype='uint8'), 8,0)

Any herlp will be very appreciated  thank you !

loicland, 2020-08-27:
Ok I am looking into it right now. What parameters did you use for:
- args.reg_strength
- args.k_nn_adj

loicland, 2020-08-27:
Ok, I solved it. the problem is that the acquisition is quite sparse (ie the scan lines have gaps), which renders the computation of the geometric features quite tricky. This can be counterbalanced by increasing the voxel_width of the pruning step. Here is what I obtain with a voxel_width of 0.2 and all other parameters as is:

[image]

and the resulting partition

[image]

to get better results I would suggest a lower reg_strength, maybe 0.05.

loicland, 2020-08-27:
Another thing I had to do, you should center the cloud to avoid float precision issue:
```
xyz = xyz - xyz.mean(0)
```

medif, 2020-08-28:
Thank you so much

closed 2020-08-28 by medif