loicland/superpoint_graph #222: make[2]: *** No rule to make target '/lib/libpython3.6m.so', needed by 'libply_c.so'.  Stop.

kcabhishek, 2020-09-02:
I keep getting this error while trying to make.

[image]

Here is my cmake.

[image]
 
Please help!

loicland, 2020-09-02:
Is your python version 3.6m? Run python to find out.

kcabhishek, 2020-09-02:
Hello @loicland. Thank you for relying so fast. My python is 3.8 and I fixed the issue by changing it in the command line.  My question is I am trying to do segmentation on a .ply file, will I be able to do that with this? If so, how?

loicland, 2020-09-02:
Could you post the question in a separate issue? And do you mean partitioning or semantic segmentation?

closed 2020-09-02 by loicland