loicland/superpoint_graph #223: Partiotioning .ply file.

kcabhishek, 2020-09-03:
I am trying to do partition on a .ply file, will I be able to do that with this? If so, how?

loicland, 2020-09-03:
You need just a few modifications:
- `partition/partition.py` line 42, if you don't want to use a train/test split
- line 136-138: comment the call to `read_las` if your cloud is a ply
- you might need to alter the function  `read_ply` in `provider.py` to read your data format.

Then run:
```
python partition/partition.py --dataset custom_dataset --ROOT_PATH data_location --voxel_width 0.05 --reg_strength 0.5
```

values for `voxel_width` should be adapted so that your cloud have more or less uniform sampling precision, and `reg_strength` depending on the desired granularity of the partition

Then use the function `visualize` with `--output_type ip` to see the results.

loicland, 2020-09-03:
It's `voxel` not `voxal`.

>  Am I in the right place?

Sure, provided that your office is similar to the ones in S3DIS, or that you have enough training data to train a model from scratch. This repo won't give you instance/object segmentation however, just the class of each points.

kcabhishek, 2020-09-04:
What kind of data do I need to train my model from scratch? Also what do you mean by just the class of each points?

loicland, 2020-09-04:
You need at least a dozen scenes with points annotated by semantic class.

The output of SPG is a class for each points. If you want object detection or instance segmentation you should use a other network. Checkout torch-points3d for example, they have most tasks implemented.

closed 2021-04-12 by loicland