loicland/superpoint_graph #224: Understanding libplc.prune() function using S3DIS dataset

Crane-YU, 2020-09-04:
Hi @loicland ,

Thank you for your wonderful work. I get lost when going through your 'prune()' function in ply_c.cpp.
Could you explain what the difference is between the input label and the pruned label, and also the pruned_object?
From my understanding, the original input labels range from [0, 13] indicating 1 backgroud + 12 objects. However, when I compile the following code, I found that the elements are not within the original range (e.g. from 0 to 13):
`if args.voxel_width > 0: `
`xyz, rgb, labels, dump = libply_c.prune(xyz.astype('f4'), args.voxel_width, rgb.astype('uint8'), labels.astype('uint8'), np.zeros(1, dtype='uint8'), n_labels, 0)`
Could you please explain `libply_c.prune`. Thank you so much.

loicland, 2020-09-14:
Hi,

sorry I missed this issue. This is a simple code that voxelized the input cloud into cubes of side `voxel_width`. The input label/object is a vector of size n_original int indicating class label/object index of each point. The pruned version are matrices of size n_pruned,n_label indicating the number of points for each label withing each cubes.

[0, 13] is 14 classes: 1 background + 13 objects. So it make sense that the pruned labels would be in  [0,13].

Crane-YU, 2020-09-17:
@loicland Thank you. While I've encountered another issue that I am using my own dataset with no label and rgb information. When I was doing 
`xyz = libply_c.prune(xyz, args.voxel_width, np.zeros(xyz.shape, dtype='u1'), np.array(1, dtype='u1'), 0)[0]`
An error occurred 
> Boost.Python.ArgumentError: Python argument types in
>     partition.ply_c.libply_c.prune(numpy.ndarray, float, numpy.ndarray, numpy.ndarray, int)
> did not match C++ signature:
>     prune(boost::python::numpy::ndarray, float, boost::python::numpy::ndarray, boost::python::numpy::ndarray, boost::python::numpy::ndarray, int, int)
It seems to me that I have to input 7 arguments while from the suggestion I can only input 5 arguments when no label or rgb information is present.
Can you explain this? Thank you.

closed 2023-06-16 by loicland