loicland/superpoint_graph #225: the miou of each category of your work on the vkitti dataset?

longmalongma, 2020-09-13:
Thank you very much for your great work ssp+spg, can you tell me the miou of each category of your work on the vkitti dataset? If you can tell me, I will cite your paper and compare it with your work in our experiment. Looking forward to your reply.

loicland, 2020-09-14:
I must have this somewhere, but weren't you able to run SPG on vkitti with our pretrained weights? Using the evaluate funcntion it should be straightforward to get these values.

longmalongma, 2020-09-15:
> I must have this somewhere, but weren't you able to run SPG on vkitti with our pretrained weights? Using the evaluate funcntion it should be straightforward to get these values.

Yes. I have run your code before, but Now I don't have a machine to run your pre-training model. Could you tell me the relevant data directly?

loicland, 2020-09-15:
Ok. Using the trained weights available on the project, I have:

```
Overall accuracy : 87.77 %
Mean accuracy    : 72.16 %
Mean IoU         : 57.65 %

         Classe :   IoU
        Terrain :  73.51 % 	 6.2e+06 points
           Tree :  89.25 % 	 1.0e+07 points
     Vegetation :  69.09 % 	 1.5e+06 points
       Building :  62.26 % 	 2.2e+06 points
           Road :  84.85 % 	 9.3e+06 points
      GuardRail :  88.20 % 	 4.2e+05 points
    TrafficSign :  39.83 % 	 2.6e+05 points
   TrafficLight :  36.01 % 	 1.1e+05 points
           Pole :  18.26 % 	 9.6e+04 points
           Misc :   7.62 % 	 4.2e+05 points
          Truck :  33.39 % 	 3.9e+05 points
            Car :  85.12 % 	 3.2e+06 points
            Van :  62.10 % 	 3.4e+05 points
```

Hope this helps!

closed 2020-09-15 by longmalongma