loicland/superpoint_graph #227: NameError: name 'Program' is not defined

zcy1065670505, 2020-10-12:
Hi ，Thank for your beautiful work on pointcloud semantic segmentation.When I try to run your code about SPG, I successfully in compiling and Sem3d data processing.But ouccr error when I try to train  the network based on Sem3d.commands are as follows:
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH /media/swjtu-rs/工作/ZCY/SEMANTIC3d --db_test_name testred --db_train_name trainval --epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --pc_attrib xyzrgbelpsv --odir "results/sema3d/trainval_best" 

Train dataset: 15 elements - Test dataset: 4 elements - Validation dataset: 0 elements
Epoch 0/500 (results/sema3d/trainval_best):
  0%|                                      | 0/7 [00:01<?, ?it/s]
Traceback (most recent call last):
  File "learning/main.py", line 459, in <module>
    main()
  File "learning/main.py", line 329, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 203, in train
    outputs = model.ecc(embeddings)
  File "/home/swjtu-rs/anaconda3/envs/pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 477, in __call__
    result = self.forward(*input, **kwargs)
  File "/media/swjtu-rs/工作/ZCY/superpoint_graph-ssp-spg/learning/../learning/graphnet.py", line 97, in forward
    input = module(input)
  File "/home/swjtu-rs/anaconda3/envs/pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 477, in __call__
    result = self.forward(*input, **kwargs)
  File "/media/swjtu-rs/工作/ZCY/superpoint_graph-ssp-spg/learning/../learning/modules.py", line 176, in forward
    self._edge_mem_limit)
  File "/media/swjtu-rs/工作/ZCY/superpoint_graph-ssp-spg/learning/../learning/ecc/GraphConvModule.py", line 80, in forward
    ctx._degs_gpu.narrow(0, startd, numd))
  File "/media/swjtu-rs/工作/ZCY/superpoint_graph-ssp-spg/learning/../learning/ecc/cuda_kernels.py", line 125, in conv_aggregate_fw
    function, stream = get_kernel_func('conv_aggregate_fw_kernel_v2', conv_aggregate_fw_kernel_v2(), get_dtype(src))
  File "/media/swjtu-rs/工作/ZCY/superpoint_graph-ssp-spg/learning/../learning/ecc/cuda_kernels.py", line 38, in get_kernel_func
    prog = Program(ksrc, kname+dtype+'.cu')        
NameError: name 'Program' is not defined

Could you tell me what's the problem?  Thank you for your kind help!

loicland, 2020-10-12:
Hi,

check line line 38 in  `/learning/ecc/cuda_kernels.py` there are instructions on how to fix this issue.

zcy1065670505, 2020-10-12:
Hi ,
I have solved this problem this morning. And successfully trained this Network by commonds:
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH /media/swjtu-rs/工作/ZCY/SEMANTIC3d --db_test_name testred --db_train_name trainval --epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --pc_attrib xyzrgbelpsv --odir "results/sema3d/trainval_best"

But I found the test test acc is too low :
-> Test Loss: 10.8944  Test accuracy: 1.28%  Test oAcc: 0.00%  Test avgIoU: nan%

loicland, 2020-10-12:
Please keep your issues focused.

This one is easy though, there are no ground truth for semantic3d for the reduced test set. You must write the label file with `write_Semantic3` and submit it on the semantic3d.net servers.

zcy1065670505, 2020-10-12:
I'm very appreciate for your repidly reply. there are several questions need your help:

No1. It means that if I use the  commands  below to train the Network based on Sem3D dataset. The test acc is based on test_full set or reduced test set?  And Because of lacking ground truth(both of est_full set and reduced test set),so the acc is too low.
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH /media/swjtu-rs/工作/ZCY/SEMANTIC3d --db_test_name testred --db_train_name trainval --epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --pc_attrib xyzrgbelpsv --odir "results/sema3d/trainval_best"

No2. If I use commands below to train the network,the acc is validition acc?
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --epochs 450 --lr_steps '[350, 400]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --pc_attrib xyzrgbelpsv --ptn_nfeat_stn 11 --nworkers 2 --odir "results/sema3d/best".

No3. The acc in your paper is which acc?

Thank you!

loicland, 2020-10-12:
1) ` --db_test_name testred` -> it is the reduced test
The ACC is not too low, it is nonsensical since there are no ground truth to compute it.

2) I think you should add `--use_val_set 1`
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --epochs 450 --use_val_set 1 --lr_steps '[350, 400]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --pc_attrib xyzrgbelpsv --ptn_nfeat_stn 11 --nworkers 2 --odir "results/sema3d/best".

If the network is already trained, add `--resume RESUME` and make sure that the `odir` contains the trained model.

3) the results in the paper are from the semantic3d.net submission servers:
http://www.semantic3d.net/view_results.php?chl=1

zcy1065670505, 2020-10-12:
Thank you so much!
It works.

closed 2020-10-12 by loicland