loicland/superpoint_graph #229: a lot of problems when compiling cmake

Fanyan-0411, 2020-11-23:
I really appreciate for your kind response first of all!!
I got some problems when doing CMAKE, here are some details of my issue. 
After doing cmake
 
[image]

Got 
[image]

It looks no bug in there, but when doing “make”, I got:
 
[image]

And I am sure boost and eigen3 are all well installed, please look:
 
[image]

The versions of cmake(3.10), boost(1.67 or 1.65), eigen3(3.3.7) and numpy(1.19) are all appropriate. Please tell me why it still can’t work, and I have tried this for a whole week. Thank you sooooooo much!!!

loicland, 2020-11-23:
Hi,

the CMAKE output indicates that like your boost libraries are not linking properly.

> Boost LIBRARIES ARE

Try 

`sudo apt-get install  libboost-all-dev`

And when running make add `VERSBOSE=1` the error is a bit generic here

Fanyan-0411, 2020-11-25:
Thanks for your replay~~~
I have fixed this issue, and the main reason could be dependency problem. The solution of the issue is shown as follows. Hoping it can give some clue to others. 
Firstly, please be sure all the packages are installed correctly. 
[image]
and then, execute command as usual

> CONDAENV=YOUR_CONDA_ENVIRONMENT_LOCATION 

> cd partition/ply_c

> cmake . -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3

[image]

in this part if you get 

> Boost LIBRARIES ARE
rather than
> Boost LIBRARIES ARE: /home/fanyan2/anaconda3/envs/pytorch3.7/lib

which means cmake didn't find **Boost LIBRARIES**. 

However **libboost-all-dev** was also installed, then you can try to SET the **Boost LIBRARIES DIR** in CMakeLists.txt in the front of command **INCLUDE_DIRECTORIES(${Boost_INCLUDE_DIRS})**, for instance
[image]

then i get 
[image]

execute make, 
[image]

emmmmmm, a lot of warnings, but ignore it.

now enjoy coding time~~

closed 2021-04-12 by loicland