loicland/superpoint_graph #231: some problem about test oAcc and  test mIoU on Semantic3D dataset

sicily373, 2020-12-06:
hi,thanks for your great work! i have trained the model on Semantic3d dataset follow your step,but met some problem. 
while i run CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --db_test_name testred --db_train_name trainval \--epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 \--nworkers 2 --pc_attrib xyzrgbelpsv --odir "results/sema3d/trainval_best".  There is a error that shows /learning/../learning/metrics.py:70: RuntimeWarning: invalid value encountered in true_divide return sum(values) / class_seen  -> Multisample 10: Test accuracy: 1.4567216464749677, 	Test oAcc: 0.0, 	Test avgIoU: nan, 	Test mAcc: 0.0.
i saw the #208 and #170, they have the same problem,and you proposed two solutions.
First, get the test sets label, but where should i get this? after getting it, how should i use it to get specific test performance?
secondly, use the train/val split, i run the CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --epochs 450 --lr_steps '[350, 400]' --test_nth_epoch 100 \ --model_config 'gru_10,f_8' --pc_attrib xyzrgbelpsv --ptn_nfeat_stn 11 --nworkers 2 --odir "results/sema3d/best", but the result is exactly the same as the previous one,and no specific test result data can be obtained.
can you please help me to answer these doubts,i will appreciate it.

loicland, 2020-12-10:
Hi,

the test labels for testred and testfull are withheld by semantic3d. If you want to evaluate your performance you should do a submission on their website.

To use the train split, set `--use_val_set` to 1.

sicily373, 2020-12-10:
thank you so much for your reply

closed 2023-06-16 by loicland