loicland/superpoint_graph #234: Test on my custom_dataset

zcy1065670505, 2020-12-21:
HI, 
When I use SPG to test my custom dataset,Do I need to make any other changes except what you mentioned in readme.
In fact, I have tested the custom dataset (LIDAR highway scene) according to your instructions, but there is always one class missing in the test results (the lane line is missing). I do this. 
First of all, my data does not have RGB(only x y z intensity). I replace the RGB channel with the intensity of lidar (copy three times),then I use the intensity as additional information besides geometric information for partition. 
Secondly, my datasets are all labeled, and there are no unlabeled points,what other changes do I need to make to get the correct segmentation results.
Thank you!
the gt and pred are as follows:
[image]
[image]

loicland, 2020-12-21:
- even if you don't have any un-annotated points, the label `0` should be reserved for un-labelled data. Start at `1`.
- replicating the intensity 3 times to mimic RGB is obviously not ideal. dapt [here](https://github.com/loicland/superpoint_graph/blob/497c52e5a6b6253673c4e8585369a46522325b91/partition/partition.py#L170), and change `--pc_attribs` and `--ptn_nfeat_stn` to only have one radiometric information.

zcy1065670505, 2020-12-24:
- Thank your helpful suggestions！ it works.

- But the traffic signs line still keeps bad result,only 0.00035 IoU, I used intensity(original intensity of my custom dataset belongs to [0,65535], I change them to [0,1] just likes  rgb/255.0) for partition, but it does not work(there are no any traffic sign lines in partition.ply or Geo.ply ),I don't konw why.

- And May I ask your a simply question.emm..., How to make the GT and Pred at the  COMPLETE SAME perspective ,use some software or some open source project? Thank you.

closed 2021-03-22 by zcy1065670505

Xiaofei-Kevin-Yang, 2021-05-12:
> * Thank your helpful suggestions！ it works.
> * But the traffic signs line still keeps bad result,only 0.00035 IoU, I used intensity(original intensity of my custom dataset belongs to [0,65535], I [...]

Hi, Thanks for your exploration on SPG. May I ask what accuracy do you get in your own dataset.