loicland/superpoint_graph #235: Resume my custom dataset of scannet failed for pytorch_geometry

ruomingzhai, 2020-12-27:
I try to train the scannet datatset in custom_dataset mode , it goes well in the first 20 epoch without pyg,and collpases at 30 epoch.And when I try to continue the training with --RESUME option , it comes out a bug like the following:
`Traceback (most recent call last):
  File "learning/main.py", line 463, in <module>
    main()
  File "learning/main.py", line 168, in main
    model = create_model(args, dbinfo)
  File "learning/main.py", line 427, in create_model
    model.ecc = graphnet.GraphNetwork(args.model_config, nfeat, [dbinfo['edge_feats']] + args.fnet_widths, args.fnet_orthoinit, args.fnet_llbias,args.fnet_bnidx, args.edge_mem_limit, use_pyg = args.use_pyg, cuda = args.cuda)
  File "/mnt/superpoint_graph_zrm/learning/../learning/graphnet.py", line 78, in __init__
    gconv = RNNGraphConvModule(cell, fnet, nfeat, vv = vv, nrepeats=nrepeats, cat_all=cat_all, edge_mem_limit=edge_mem_limit, use_pyg = use_pyg, cuda = cuda)
  File "/mnt/superpoint_graph_zrm/learning/../learning/modules.py", line 145, in __init__
    self.nn = NNConv(nfeat, nfeat, vv = vv)
NameError: name 'NNConv' is not defined`
This bug makes no sense because it went well in the first 30 epoch and I did not change any code during this training process.
Can anyone give me some clues? Thanks!

loicland, 2020-12-28:
Hi,

do you have [PyTorch Geometric](https://pytorch-geometric.readthedocs.io/en/latest/) installed on your computer?

ruomingzhai, 2020-12-28:
> Hi,
> 
> do you have [PyTorch Geometric](https://pytorch-geometric.readthedocs.io/en/latest/) installed on your computer?

I did not install the pytorch geometric because I thought it is optional choice.my programmes went well without pytorch geometric until the 30th epoch.I don't know what is going on.

loicland, 2020-12-28:
Hi,

change the default value of `use_pyg` [here](https://github.com/loicland/superpoint_graph/blob/497c52e5a6b6253673c4e8585369a46522325b91/learning/modules.py#L134) from `True` to `False`

loicland, 2020-12-28:
can you check the value of args.use_pyg just [after](https://github.com/loicland/superpoint_graph/blob/497c52e5a6b6253673c4e8585369a46522325b91/learning/main.py#L396) the checkpoint is loaded?

`print(args.use_pyg )`

ruomingzhai, 2020-12-29:
> can you check the value of args.use_pyg just [after](https://github.com/loicland/superpoint_graph/blob/497c52e5a6b6253673c4e8585369a46522325b91/learning/main.py#L396) the checkpoint is loaded?
> 
> `print(args.use_pyg )`

[image]
and the program works now, I found that this "if use_pyg:" return true anyway no matter what the value of use_pyg is.Because I add the print(args.use_pyg) you recommand and it shows 0,but the code inside the "if use_pyg:" still be excuted

ruomingzhai, 2020-12-29:
> can you check the value of args.use_pyg just [after](https://github.com/loicland/superpoint_graph/blob/497c52e5a6b6253673c4e8585369a46522325b91/learning/main.py#L396) the checkpoint is loaded?
> 
> `print(args.use_pyg )`

and the program works now, I found that this "if use_pyg:" return true anyway no matter what the value of use_pyg is.Because I add the print(args.use_pyg) you recommand and it shows 0,but the code inside the "if use_pyg:" still be excuted.So I change and add two lines as following:
`        self.use_pyg = False
        use_pyg=False`
in modules.py in line 143 & 144

ruomingzhai, 2020-12-30:
> and the program works now, I found that this "if use_pyg:" return true anyway no matter what the value of use_pyg is.Because I add the print(args.use_pyg) you recommand and it shows 0,but the code [...]

> and the program works now, I found that this "if use_pyg:" return true anyway no matter what the value of use_pyg is.Because I add the print(args.use_pyg) you recommand and it shows 0,but the code [...]

> can you check the value of args.use_pyg just [after](https://github.com/loicland/superpoint_graph/blob/497c52e5a6b6253673c4e8585369a46522325b91/learning/main.py#L396) the checkpoint is loaded?
> 
> `print(args.use_pyg )`

I install pyg and the problem is sloved, thanks for your guidance!

closed 2020-12-30 by ruomingzhai

kangx326, 2022-09-18:
Hi, can you share your implementation on ScanNet-v2? I'm working on this right now and need some help.

kangx326, 2022-09-18:
> > Hi,
> > do you have [PyTorch Geometric](https://pytorch-geometric.readthedocs.io/en/latest/) installed on your computer?
> 
> I did not install the pytorch geometric because I thought it is optional choice.my programmes went well without pytorch geometric until the 30th epoch.I don't know what is going on.

Hi, can you share your implementation on ScanNet-v2? I'm working on this right now and need some help.

loicland, 2022-09-18:
In the end, we never got around to implement it.

This [repo](https://github.com/fpthink/SPNet) use a superpoint approach on ScanNet, maybe it will help you.