loicland/superpoint_graph #239: Compile the libply_c and libcp libraries

zouhair-geo, 2021-03-11:
Hello Sir, 
Can you help me please I always receive errors( errors in the paths) in this part although I replaced $CONDAENV by the path of my environment (C:\ProgramData\Anaconda3 ) and also I do not have the same files for example at the level of Lib I do not have any file which is called libpython3.6.so 
I am working with python 3.8
Thank you in advance

loicland, 2021-03-12:
Have you tried replacing mentions of python3.6 by python3.8 in the command lines?

zouhair-geo, 2021-03-12:
Yes Professor I replaced 3.6 with 3.8 but in Lib I don't have any file called libpython3.8.so

loicland, 2021-03-12:
If you have installed conda correctly there must be a `libpython*.so` file in CONDAENV/lib. On a UNIX terminal type:

`libpy="lib/libpython*.so"`
`locate $CONDAENV$libpy

zouhair-geo, 2021-03-12:
I don't use Unix. I have installed Anaconda (Python 3.8) on Windows 10. However I don't find the files you said.
Thank you

loicland, 2021-03-12:
I don't use windows, so I am not sure how to compile the librairies, but I know it is possible. Have you installed `python3-dev`? Anyway, first step for you would be to locate or install `libpython*.so`.

zouhair-geo, 2021-03-12:
thank you very much professor

closed 2023-06-16 by loicland