loicland/superpoint_graph #241: Question about the partition step?

linlinge, 2021-03-12:
Q1:
In the partition step, a 10-nearest neighbor adjacent graph is established at first. But the paper do not mention how to set the edge weight, and I find it in the code that is demonstrated in the following:
[image]
I am confused about this formula, why not using 1/distances ?

Q2：
when I use the l0-cut pursuit algorithm, it was not work. Is there anything that I should be pay attention, for an example, some parameters?

loicland, 2021-03-12:
Q1 - 1/d would be unstable for very small distances, and cause numerical errors for doubled points. Furthermore, the weights would not be unitless, and changing from cm to inches for example would change the results. 1/(lambda + d/d0) is unitless and more stable. However this is a heuristics and I am sure there are better ways to do it.

Q2 - what error are you receiving?

linlinge, 2021-03-12:
Q1:
Thank you very much! This explanation inspired me a lot. 

Q2:
For the l0-cut algorithm, I try it with another example. Look at the figure bellow
[image]
This is a simple point cloud with 18 points. And the segment on each point is its corresponding normal. If I use the normal as the feature, and aim to partition this point cloud into two parts as bellow: 
[image]
I write the code in the bellow:
[image]
Here, feature1 stores the normals and is a nx3 matrix, n is the number of points (n=18). Then l0-cut pursuit only returns one cluster.
Then I change the feature1 to feature2
[image]
Then I got an answer like this
[image]
This result is not as expected. It seems l0-cut pursuit is sensitive to the dimension of the feature. Besides, when I apply it to a more complex point cloud, it returns a result with one cluster. So, where I do wrong, feature design, parameter tuning or something else?

loicland, 2021-03-12:
could you send me an archive with the `features`, `source`, `target` as .npy for your toy example? I will take a look as soon as I find a few minutes.

linlinge, 2021-03-13:
You can download them in this link:
https://drive.google.com/drive/folders/14TPQuUJIE7-AOP9Zg_mz3gX7yLzA0Ikb?usp=sharing

closed 2023-06-16 by loicland