loicland/superpoint_graph #243: Pandas Tokenizing Error in supervized_partition/graph_processing.py

mati3230, 2021-03-16:
Hi @all, 
great project! I wanted to test the feature "SSP+SPG: SPG with learned partition". To do so, I executed the fix "cp S3DIS_fix.diff $S3DIS_DIR/data; cd $S3DIS_DIR/data; git apply S3DIS_fix.diff; rm S3DIS_fix.diff; cd -" from the file [S3DIS.md](https://github.com/loicland/superpoint_graph/blob/ssp%2Bspg/S3DIS.md) which seems to work fine. After that, I executed the command "python supervized_partition/graph_processing.py --ROOT_PATH $S3DIS_DIR --dataset s3dis --voxel_width 0.03". Now the following Stacktrace appears while processing Area 3:

=================
   Area_3/
=================
1 / 23---> storage_2
    graph structure already computed - delete for update...
2 / 23---> office_3
    graph structure already computed - delete for update...
3 / 23---> conferenceRoom_1
    graph structure already computed - delete for update...
4 / 23---> lounge_1
    graph structure already computed - delete for update...
5 / 23---> office_1
    graph structure already computed - delete for update...
6 / 23---> office_10
    graph structure already computed - delete for update...
7 / 23---> hallway_3
    graph structure already computed - delete for update...
8 / 23---> office_5
    graph structure already computed - delete for update...
9 / 23---> WC_1
    graph structure already computed - delete for update...
10 / 23---> hallway_4
    graph structure already computed - delete for update...
11 / 23---> office_4
    graph structure already computed - delete for update...
12 / 23---> lounge_2
    graph structure already computed - delete for update...
13 / 23---> office_2
    graph structure already computed - delete for update...
14 / 23---> office_6
    graph structure already computed - delete for update...
15 / 23---> office_8
    graph structure already computed - delete for update...
16 / 23---> office_9
    graph structure already computed - delete for update...
17 / 23---> hallway_1
    graph structure already computed - delete for update...
18 / 23---> WC_2
    graph structure already computed - delete for update...
19 / 23---> hallway_6
    graph structure already computed - delete for update...
20 / 23---> office_7
    graph structure already computed - delete for update...
21 / 23---> storage_1
    graph structure already computed - delete for update...
22 / 23---> hallway_2
    computing graph structure...
Traceback (most recent call last):
  File "supervized_partition/graph_processing.py", line 562, in <module>
    main()
  File "supervized_partition/graph_processing.py", line 121, in main
    xyz, rgb, labels, objects = read_s3dis_format(data_file)
  File "/media/mati3230/d8196a77-44d0-4208-a7d3-713df9c812f5/projects/superpoint_graph/supervized_partition/../partition/provider.py", line 189, in read_s3dis_format
    room_ver = pd.read_csv(raw_path, sep=' ', header=None).values
  File "/home/mati3230/miniconda3/envs/landrieu/lib/python3.6/site-packages/pandas/io/parsers.py", line 688, in read_csv
    return _read(filepath_or_buffer, kwds)
  File "/home/mati3230/miniconda3/envs/landrieu/lib/python3.6/site-packages/pandas/io/parsers.py", line 460, in _read
    data = parser.read(nrows)
  File "/home/mati3230/miniconda3/envs/landrieu/lib/python3.6/site-packages/pandas/io/parsers.py", line 1198, in read
    ret = self._engine.read(nrows)
  File "/home/mati3230/miniconda3/envs/landrieu/lib/python3.6/site-packages/pandas/io/parsers.py", line 2157, in read
    data = self._reader.read(nrows)
  File "pandas/_libs/parsers.pyx", line 847, in pandas._libs.parsers.TextReader.read
  File "pandas/_libs/parsers.pyx", line 862, in pandas._libs.parsers.TextReader._read_low_memory
  File "pandas/_libs/parsers.pyx", line 918, in pandas._libs.parsers.TextReader._read_rows
  File "pandas/_libs/parsers.pyx", line 905, in pandas._libs.parsers.TextReader._tokenize_rows
  File "pandas/_libs/parsers.pyx", line 2042, in pandas._libs.parsers.raise_parser_error
pandas.errors.ParserError: Error tokenizing data. C error: Expected 6 fields in line 926337, saw 7

Is there any step I have not taken?

My S3DIS-DIR variable is as follows: S3DIS_DIR="/path/to/S3DIS/Stanford3dDataset_v1.2_Aligned_Version". This directory contains a "data"-folder. The "data"-folder contains the area data, e.g. "/path/to/S3DIS/Stanford3dDataset_v1.2_Aligned_Version/data/Area_1".

I use Ubuntu 20.04 as operating system. Furthermore, I use a miniconda environment in which I installed the python dependencies. 

Best regards, Marcel

loicland, 2021-04-12:
Hi,

sorry for delayed response. I would start by inspecting manually the incriminating fil and line.

mati3230, 2021-05-19:
Thank you for your answer -- manually fixing the line seems to do the job.

closed 2021-05-19 by mati3230