loicland/superpoint_graph #246: problem with the subsample of super point for gru training

ruomingzhai, 2021-04-18:
i have a question about the subgraph.your paper says the total number of super points in each batchsize was Subsampp to 512 (derived from --args.hardcutoff).I am not sure it was before embedding or after the embedding(i.e. in graphconv).Because I print the size of embeddind output in batchsize =2,it is Tensor.size(1174，32)，a little bit larger than 512*2=1024.Can you explain and point out to me where exactly is the subsample code.It bothers me long time!Thx!

loicland, 2021-04-19:
Is this during training or inference?

ruomingzhai, 2021-04-19:
> Is this during training or inference?

During training

loicland, 2021-04-19:
Ah yes. The hard cutoff is the maximum number of "valid" superpoints to embed with the CloudEmbedder. A superpoint is valid if it has more than `args.ptn_minpts` points. Indeed, superpoints with too few points are attributed an all zero embedding at no computational cost.

In your batch you have: 1024 spoints with > 40pts, and 50 with 1<40 points.

See the following lines for the details:
`/learning/spg.py#L143`
`/learning/spg.py#L123`
`/learning/pointnet.py#L149`

ruomingzhai, 2021-04-21:
> Ah yes. The hard cutoff is the maximum number of "valid" superpoints to embed with the CloudEmbedder. A superpoint is valid if it has more than `args.ptn_minpts` points. Indeed, superpoints with too few points are attributed an all zero embedding at no computational cost.
> 
> In your batch you have: 1024 spoints with > 40pts, and 50 with <40 points.
> 
> See the following lines for the details:
> `/learning/spg.py#L143`
> `/learning/spg.py#L123`
> `/learning/pointnet.py#L149`

Oh I see.why I care about the shape of embeding is I want to use your superpoint as a fixed 1024 dimension embedding to feed my own designed model.Is there anyway to get just 1024 superpoint,which reqires the ptn_hardcutoff rule.

loicland, 2021-04-21:
replacing:
```
G = k_big_enough(G, args.ptn_minpts, args.spg_augm_hardcutoff)
```
by
  ```
G = G.subgraph(range(args.spg_augm_hardcutoff))
```
should work.