loicland/superpoint_graph #247: libply_c compilation failing: fatal error: pyconfig.h: No such file or directory

sandeepnmenon, 2021-04-20:
Trying to setup the repository and failing at the 'make' step for libply_c
```
CONDAENV=YOUR_CONDA_ENVIRONMENT_LOCATION
cd partition/ply_c
cmake . -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.8m.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.8m -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3
make
```

Error
```
[ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o
In file included from /home/anaconda3/envs/superpoint/include/boost/python/detail/prefix.hpp:13:0,
                 from /home/anaconda3/envs/superpoint/include/boost/python/args.hpp:8,
                 from /home/anaconda3/envs/superpoint/include/boost/python.hpp:11,
                 from /home/SandeepMenon/superpoint/superpoint_graph/partition/ply_c/ply_c.cpp:6:
/home/anaconda3/envs/superpoint/include/boost/python/detail/wrap_python.hpp:57:11: fatal error: pyconfig.h: No such file or directory
 # include <pyconfig.h>
           ^~~~~~~~~~~~
compilation terminated.
CMakeFiles/ply_c.dir/build.make:62: recipe for target 'CMakeFiles/ply_c.dir/ply_c.cpp.o' failed
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ply_c.dir/all' failed
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

Environment details
conda 4.9.2
Python 3.8.5

loicland, 2021-04-20:
What did you set `CONDAENV` to?

sandeepnmenon, 2021-04-20:
> What did you set `CONDAENV` to?

CONDAENV=/home/anaconda3/envs/superpoint/

sandeepnmenon, 2021-04-20:
The issue was with the anaconda version of python
I gave `libpython3.8m` instead of `libpython3.8`.
The following command worked
```
cmake . -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.8.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.8 -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3
make
```

closed 2021-04-20 by sandeepnmenon