loicland/superpoint_graph #256: Question on custom dataset

HuangDihe, 2021-06-08:
I run partition.py to prune my point cloud(63825 points, xmin=-0.78, xmax=-0.0367, ymin=-0.1375, ymax=0.08109, zmin=-0.552, zmax=0.568) and i set voxel_width=0.01
and it shows:

`Voxelization into 0 x 0 x 0 grid`
`Reduced from 63825 to 1 points (0%)`

How should i fix thie?
Thanks

qzq-hit, 2021-09-05:
> I run partition.py to prune my point cloud(63825 points, xmin=-0.78, xmax=-0.0367, ymin=-0.1375, ymax=0.08109, zmin=-0.552, zmax=0.568) and i set voxel_width=0.01
> and it shows:
> 
> `Voxelization into 0 x 0 x 0 grid`
> `Reduced from 63825 to 1 points (0%)`
> 
> How should i fix thie?
> Thanks
Have you solved this problem? It happened when I was pruning my own data set

antoine-lag, 2022-02-25:
I had the same problem and it came from line 192 of ply_c.cpp, the type of xyz was not float but double

closed 2023-06-16 by loicland