loicland/superpoint_graph #260: The introduction of the ruturn of libcp.cutpursuit

shaofeifei11, 2021-08-30:
I am sorry, I can not understand what is the "components", "in_component" (lines 177 of partition/partition.py )

`components, in_component = libcp.cutpursuit(features, graph_nn["source"], graph_nn["target"]
                                         , graph_nn["edge_weight"], args.reg_strength)`

In particular, "in" in "in_component" is an abbreviation of "index"? 

Could you briefly introduce what data is stored in "component" and "In_component"? 

I am very expected to get your help.  Thank you!

azsxdcfvg, 2021-11-19:
I think that in_component is the number of components which this point is belonging to.

cama36, 2022-11-22:
hello, i have the same question, do you understand them now?

loicland, 2022-11-22:
Hi, sorry for delayed answer.

`components` is a list of size the number of components. `components[c]` is the list of the indices of the vertices in component c.

`in_component` is an array of size n_vertex. `in_component[v]` returns the index of the component containing v.

In other words, `in_component[components[c]]==c`

closed 2023-06-16 by loicland