loicland/superpoint_graph #271: cuda acceleration for partition

ZheningHuang, 2022-07-19:
Do you think it is possible to use Cuda to accelerate the speed of handcrafted partition? 

Currently, it takes 3-5 seconds to process a 0.04 voxelized file in the S3DIS dataset and I am thinking of accelerating this with CUDA. Will be appreciated if you could give us know your thoughts on this. Is it feasible and if so what would be a good way to do this.

Best,
Zhening

loicland, 2022-07-20:
Hi,

Parallelizing the current partition code with CUDA would not be easy at all.

We are working on an accelerated version, which we should release this fall and is roughly x10 faster.

Otherwise, you could use another CUDA-friendly partition scheme such as mean-field and keep the rest as is.

ZheningHuang, 2022-07-31:
Hi Loic,

Thank you so much for your reply. Could you please point me in the direction of the mean-field partition scheme which is more CUDA-friendly? Maybe get me a link to the GitHub Repo or the name of the paper.

I am testing different partition schemes, including yours for a paper and it would be really appreciated if you could help.

Best regards,
Zhening

loicland, 2022-08-02:
You can try this paper and associated repo:

https://openaccess.thecvf.com/content/ICCV2021/html/Hui_Superpoint_Network_for_Point_Cloud_Oversegmentation_ICCV_2021_paper.html

it's not mean field but some sort of soft-kmeans that is GPU friendly, provided that the input graphs are not too large.

As for parallelized cut pursuit, we will release the new version this fall.

closed 2023-06-16 by loicland