loicland/superpoint_graph #272: TypeError in 'NearestNeighbors(1, algorithm='kd_tree').fit(xyz)'

Pixie412, 2022-08-17:
Dear authors,

When I running 'python partition/partition.py', an error occurred here in the partition/provider.py: 
[image]

What is the value '1' mean in the NearestNeighbors()?

loicland, 2022-08-25:
It depends on your version of sklearn. Replace with:

`nn=NearestNeighbors(n_neighbors=1,algorithm='kd_tree').fit(xyz)`

Pixie412, 2022-08-27:
Hi,
Thanks for the reply. I encountered another problem when computing area3 hallway_2: 
[image]
I didn't modify the the original s3dis data. Neither do I found any strange character in the hallway_2 file. How to solve this error？

closed 2022-08-27 by Pixie412