loicland/superpoint_graph #273: Inconsistent class_maps for s3dis

fclairec, 2022-08-25:
Dear authors, 

Nice work done here - thanks. 
But, it seems to me as if the labels are inconsistently evaluated.. 
I have written down the steps I checked, and it seems that the inconsistency only effects the final per_class_iou dictionary. Please correct me if I am wrong.

1. in provider.py object_name_to_label() called in read_s3dis_format() will assign labels as follows 'ceiling': 1, ... ,   'clutter': 13, ... , 'stairs': 0,
2. the next step is to train the supervised partition (supervised_partition.py) where get_s3dis_info() returns 0:'ceiling' and 12:'clutter'. However dbinfo['inv_class_map'] seems not to be used here - so it does not cause issues. 
3. Also when writing the parsed files (by running s3dis_dataset.py) all seems still fine (labels of step 1 are used)
4. However then when training with main.py, the line dbinfo = s3dis_dataset.get_info(args) fetches the labels as follows 0:'ceiling' and 12:'clutter'. The Training seems fine. In eval_final(), however dbinfo['inv_class_map'].items() is used to gather the per class ious. This will cause false reporting.

loicland, 2022-08-25:
This is because `0`  is used as the nolabel class during training (stairs are not evaluated). Since the nolabel class is never predicted, we shift the labels by one when evaluating.

To see this, remember that the columns of `target` for a superpoint are as follows, (see [here](https://github.com/loicland/superpoint_graph/blob/c7924c0066ead4cf88937045855655ad3cbfdc57/learning/spg.py#L70)):
- col0:   label mode, ie the most common label the superpoint
- col1:  number of unannotated points
- col2+:  number of point of each label

When training, we use the cross entropy with the label mode (target[0]), see [here](https://github.com/loicland/superpoint_graph/blob/c7924c0066ead4cf88937045855655ad3cbfdc57/learning/main.py#L193). When evaluating, we compute the precision of the point-level predictions (target[2:]), see [here](https://github.com/loicland/superpoint_graph/blob/c7924c0066ead4cf88937045855655ad3cbfdc57/learning/main.py#L259).

fclairec, 2022-08-31:
Thanks @loicland for the explanation. I found my thinking mistake..

closed 2022-08-31 by fclairec