loicland/superpoint_graph #276: When making ply_c, fatal error: numpy/ndarrayobject.h: No such file or directory

whuhxb, 2023-02-13:
CMakeLists.txt

# Graph for Cut Pursuit
# author: Loic Landrieu
# date: 2017

CMAKE_MINIMUM_REQUIRED(VERSION 3.5)

PROJECT(LIBGEOF)

SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -std=c++11 -fopenmp -O3")

##############################
### Find required packages ###
##############################

find_package(Boost 1.65.1 COMPONENTS python numpy REQUIRED)

#include_directories(${Boost_INCLUDE_DIRS})
include_directories(${Boost_INCLUDE_DIRS} ${EIGEN3_INCLUDE_DIRS})
link_directories(${Boost_LIBRARY_DIRS})

message("Boost includes ARE " ${Boost_INCLUDE_DIRS})
message("Boost LIBRARIES ARE " ${Boost_LIBRARY_DIRS})

find_package(Eigen3 REQUIRED NO_MODULE)
INCLUDE_DIRECTORIES(${EIGEN3_INCLUDE_DIR})
#LINK_DIRECTORIES(${EIGEN3_LIBRARY_DIRS})

#SET(PYTHON_LIBRARIES /usr/lib/x86_64-linux-gnu/libpython2.7.so)
#SET(PYTHON_INCLUDE_DIRS /usr/include/python2.7)
find_package(PythonLibs)
message("PYTHON LIBRARIES ARE " ${PYTHON_LIBRARIES})
INCLUDE_DIRECTORIES(${PYTHON_INCLUDE_DIRS})
LINK_DIRECTORIES(${PYTHON_LIBRARY_DIRS})
##############################
###  Build target library  ###
##############################

set(CMAKE_LD_FLAG "${CMAKE_LD_FLAGS} -shared -Wl -fPIC --export-dynamic -fopenmp -O3 -Wall")

add_library(ply_c SHARED ply_c.cpp)
target_link_libraries(ply_c
    ${Boost_LIBRARIES}
    ${PYTHON_LIBRARIES}
    )

When make, the errors occur like this.
partition/ply_c/ply_c.cpp:8:33: fatal error: numpy/ndarrayobject.h: No such file or directory
compilation terminated.
CMakeFiles/ply_c.dir/build.make:81: recipe for target 'CMakeFiles/ply_c.dir/ply_c.cpp.o' failed
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
CMakeFiles/Makefile2:94: recipe for target 'CMakeFiles/ply_c.dir/all' failed
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2

MEIXuYan, 2023-02-26:
It shows you didn't install numpy lib or the complier can not find suitable numpy version. Given the full output log when you do `mkdir build && cd build/ && cmake .. && make` may help.

MEIXuYan, 2023-02-26:
In my case, the higher version of cmake tend to find numpy of python3.8, we can fix it in `CMakeLists.txt` file , change follow lines from

```txt
find_package(PythonLibs)
find_package(PythonInterp)
```
to

```txt
find_package(PythonLibs 3.7 REQUIRED)
find_package(PythonInterp 3.7 REQUIRED)
```
my runtime python version is 3.7, you can change it to your runtime python version.