loicland/superpoint_graph #277: How to control the number of superpoints in a room?

MEIXuYan, 2023-02-25:
Hello,

Thank you for sharing the code.  Could please show me how to control the number of superpoints in a room in both supervised and unsupervised partion?

ardianumam, 2023-03-15:
Up! I also wonder how to adjust the number of superpoints in the method.

Thanks.

loicland, 2023-03-15:
You can simply change `--reg_strength`: higher values will yield fewer superpoints.

With supervised partitions, you have more options:

--CP_cutoff determine the minimum size of a Superpoint

--spatial_emb determines the weight of the spatial extent of superpoints : high value -> superpoint mostly determined by position, low value: mostly by geometry/color.

ardianumam, 2023-03-15:
Thanks for your reply :) 
The `--CP_cutoff` and `--spatial_emb` are understandable to me. For `reg_strength`, could you elaborate more on how this value can impact the number of superpoints? I know, in [this code](https://github.com/loicland/superpoint_graph/blob/497c52e5a6b6253673c4e8585369a46522325b91/supervized_partition/losses.py#L82), it will be used in the `cut-pursuit` algorithm. But I still don't know intuitively how this hyperparameter will impact to the superpoints number in the `cut-pursuit` algorithm. Thanks!

loicland, 2023-03-15:
Cut pursuit approximates a signal on a graph with simple (short) contour. To do so, we penalize a fidelity loss with the weight of the cut between constant component **multiplied by `reg_strength`**. To simplify, this `reg_strength` expresses how much *approximation error* is allowed per *unit of contour* (cross edges). In our case, since the graph signal approximated is a set of geometric/radiometric features , it rules the tradeoff between the superpoint geometric simplicity and their shapes. 

Small `reg_strength`:  the contour penalty is not very important, will yield many small superpoint with very simple geometry.

Large `reg_strength`:  the contour penalty dominates, will yield few superpoint with very simple shape.

In practice, the relationship between `reg_strength` and the number of superpoints is complicated and depends on many things (point density, sensor precision, what kind of shapes you are using, etc.). The easiest way would be to try a few values on a sample scene until the smallest object of interest are in their own superpoint.

MEIXuYan, 2023-03-15:
@loicland Thank you for your reply.
I have read the ssp papar and found the regularization strength definition(Supplementary Material A.1) in the GPMP equation, thank you for your explanation again. But I am still confusion about how the `--spatial_emb` can influence the number of superpoints? In my understanding the supervised LPE generate the main feature for cut pursuit algorithm.

closed 2023-06-16 by loicland