loicland/superpoint_graph #279: Running on Stanford3dDataset_v1.2_Aligned_Version, the error occurs.

whuhxb, 2023-03-09:
Have you ever met this error when running partition.py? Thanks a lot.

Timer : 100.8 / 1031.9 / 806.3
6 / 23---> hallway_2
    creating the feature file...
Traceback (most recent call last):
  File "partition/partition.py", line 123, in <module>
    xyz, rgb, labels = read_s3dis_format(data_file)
  File "/export/home/hanxiaobing/Documents/PlaneNet_PlaneRCNN/DGCNN_PointNet2/SensatUrban/SPG/superpoint_graph-release/partition/provider.py", line 163, in read_s3dis_format
    nn = NearestNeighbors(n_neighbors=2, radius=0.4).fit(xyz)  # added by hxb
  File "/export/home/hanxiaobing/anaconda3/envs/SPG/lib/python3.7/site-packages/sklearn/neighbors/_unsupervised.py", line 166, in fit
    return self._fit(X)
  File "/export/home/hanxiaobing/anaconda3/envs/SPG/lib/python3.7/site-packages/sklearn/neighbors/_base.py", line 435, in _fit
    X = self._validate_data(X, accept_sparse="csr")
  File "/export/home/hanxiaobing/anaconda3/envs/SPG/lib/python3.7/site-packages/sklearn/base.py", line 566, in _validate_data
    X = check_array(X, **check_params)
  File "/export/home/hanxiaobing/anaconda3/envs/SPG/lib/python3.7/site-packages/sklearn/utils/validation.py", line 800, in check_array
    _assert_all_finite(array, allow_nan=force_all_finite == "allow-nan")
  File "/export/home/hanxiaobing/anaconda3/envs/SPG/lib/python3.7/site-packages/sklearn/utils/validation.py", line 116, in _assert_all_finite
    type_err, msg_dtype if msg_dtype is not None else X.dtype
ValueError: Input contains NaN, infinity or a value too large for dtype('float32').

loicland, 2023-03-09:
Did you apply the fix on S3DIS?

whuhxb, 2023-03-09:
> Did you apply the fix on S3DIS?

After conducting fix on S3DIS, the error still occurs.

Timer :  98.6 / 1019.4 / 794.3
6 / 23---> hallway_2
    creating the feature file...
Traceback (most recent call last):
  File "partition/partition.py", line 123, in <module>
    xyz, rgb, labels = read_s3dis_format(data_file)
  File "/export/home/hanxiaobing/Documents/PlaneNet_PlaneRCNN/DGCNN_PointNet2/SensatUrban/SPG/superpoint_graph-release/partition/provider.py", line 155, in read_s3dis_format
    room_ver = genfromtxt(raw_path, delimiter=' ')
  File "/export/home/hanxiaobing/anaconda3/envs/SPG/lib/python3.7/site-packages/numpy/lib/npyio.py", line 2124, in genfromtxt
    raise ValueError(errmsg)
ValueError: Some errors were detected !
    Line #926337 (got 7 columns instead of 6)

Have you ever met this error? Thanks a lot.

loicland, 2023-03-09:
maybe check line 926337 in your hallway_2 file, apparently there a missing character, you can fix it by hand.

whuhxb, 2023-03-09:
> maybe check line 926337 in your hallway_2 file, apparently there a missing character, you can fix it by hand.

which area?

whuhxb, 2023-03-09:
> 926337

OK. Checked. Thanks.

closed 2023-03-09 by loicland